PaulStoffregen/cores #721: Teensy 4.x semi-bricked with "smallest code" on Arduino IDE 2.x

dronus, 2023-09-25:
When building for Teensy 4.x, the Arduino IDE 2.1.1 or 2.2.0 set to "smallest code" optimization produces crashing binaries.

If flashed to the device, the Teensy bricks and does not provide an USB device anymore.

The bootloader button is still working and can be used to reflash the device. 
When flashed with "fast", "faster" or "fastest" code afterwards, the device is working as usual again.

However, there is some need as there are projects that won't fit Teensy 4.x on "fast" optimized code.

freemin7, 2024-08-02:
This seems really hard to reproduce or address without a (small) code example. I am running Arduino IDE 2.3.2 and never had a binary crash on me. Does your problem persist for newer Arduino IDE versions?

SpencerLommel, 2025-04-09:
> This seems really hard to reproduce or address without a (small) code example. I am running Arduino IDE 2.3.2 and never had a binary crash on me. Does your problem persist for newer Arduino IDE versions?

I agree, this can't really be fixed without more info or a code example. I tested around with the smallest code option for a few hours yesterday and couldn't reproduce this bug either.

h4yn0nnym0u5e, 2025-04-09:
This may have more to do with the Teensyduino version than the Arduino IDE. A bunch of fixes were put into Teensyduino 1.59, which was released February 2024, i.e. after this issue was opened. _However_, my experience is that those changes do _not_ fix all circumstances in which the Teensy can fail to start up correctly. I can trigger it _intermittently_ by trying to use the TeensyDebug and Audio libraries in the same sketch, with un-modified Teensyduino 1.59. I have an empirical fix PR #673 which works for me, and have put a plea into the 1.60 beta thread that it gets some consideration. If I get any traction on this I'll happily attempt to create a small reproducible example.

If @dronus is still around 18 months later, and can show my fix is effective, that will be another data point in favour of my PR.

See https://forum.pjrc.com/index.php?threads/teensy4-1-optimisation-0g-00-fails-to-start.71846/