PaulStoffregen/cores #726: Teensy 3 includes new.h, decalring cxa_guard... even with gcc11

swanah, 2023-11-15:
With the update to the new version / compiler version we receive the following error:

```
...\framework-arduinoteensy\cores\teensy3/new.h:46:16: error: conflicting declaration of C function 'int cxa_guard_acquire(__guard*)'
```

I am using PlatformIO for compilation. The platform version is teensy@4.18.0. While testing I have also replaced framework/cores with the current HEAD from github. While this solved other long standing problems, it did not solve the error above.
I also looked at the code here and I see that for Teensy3 Printable.h still includes new.h leading to the above error message.

Am I correct, that the guard definitions in new.h are no longer required with the new compiler?

PaulStoffregen, 2023-11-17:
I can't help with PlatformIO.

I can tell you the core library was updated in several ways to use to the new toolchain.  Maybe you're got those mismatched?  I'd recommend compiling your program with Arduino IDE to check.

swanah, 2023-11-17:
Dear Paul, I am sorry, but the current setup of our project is not easily allowing to switch to ArduinoIDE. 
However, since I cannot easily compile in ArduinoIDE, I have downloaded and installed the latest ArduinoIDE + Teensyduino and compared the packages installed (i.e. the core library and the teensy tools) with those we use in our PlatformioIO setup on a file by file basis. All files are identical and I am quite certain I am using the tools and libaries consistent with this repository and teensyduino.

Some more details to our issue that might help understanding:
Our issue is triggered by functions initializing local static variables which we use in the context of implementing lazy initialized singletons. This worked in the previous Teensyduino versions including 1.57. It started throwing the error when updating to 1.58.

Looking more closely I see, that the arm-gcc in teensyduino brings along its own definition of __cxa_guard functions in libstdc++. Hence the error when the complier finds the redefinition in teensy3/new.h. 

Personally I can mitigate the problem by either removing the definitions in new.h or by compiling with "-fno-threadsafe-statics".

However, I felt it might be useful to point to this problem, as I had the impression, it is an issue in the cores introduced by upgrading the compiler version.

PaulStoffregen, 2023-11-17:
Oh, sounds like the problem is resolved, right?

swanah, 2023-11-20:
I managed to create a small example [(here on github)](https://github.com/swanah/teensy36_guard_error) that shows the problem when using ArduinoIDE comiling for *Teensy3.6*.
Doing so I realized 2 things:
- it works for Teensy 4.0 / 4.1 because -fno-threadsafe-statics is use by default (as opposed to Teensy3.6)
- the issue is itroduced, when multiple cpp-files have to be compiled and these compilation units are not explicitly including "Arduino.h"

So in a way the issue seems to be resolved, at least for me right now.
Thanks for your help and time.