PaulStoffregen/cores #736: FastLED (v3.6.0) +  WS2812Serial Broken in Teensyduino 1.59?

gfvalvo, 2024-02-26:
Teensyduino 1.59 ships with FastLED 3.6.0. This combination doesn't work with the WS2812Serial. The code hangs on the call to "pLed->init();" in the addLeds() function of FastLED.cpp:

```
CLEDController &CFastLED::addLeds(CLEDController *pLed,
								  struct CRGB *data,
								  int nLedsOrOffset, int nLedsIfOffset) {
	int nOffset = (nLedsIfOffset > 0) ? nLedsOrOffset : 0;
	int nLeds = (nLedsIfOffset > 0) ? nLedsIfOffset : nLedsOrOffset;

	pLed->init();
	pLed->setLeds(data + nOffset, nLeds);
	FastLED.setMaxRefreshRate(pLed->getMaxRefreshRate(),true);
	return *pLed;
}
```

To reproduce the issue, simply try the "FastLED_Cylon.ino" example that ships with Teensyduino. My setup was Arduino 1.8.19 + Teensyduino 1.59 + FastLED 3.6.0 on a T3.6 baord.

I first posted this as a question on the Forum [here](https://forum.pjrc.com/index.php?threads/is-fastled-ws2812serial-broken-in-teensyduino-1-59.74548/).

Paul has already reproduced the problem.

gfvalvo, 2024-03-13:
So as far as I can tell, it's not just WS2812Serial and FastLED (v3.6.0) that are broken with Teensyduino 1.59. Rather Teensyduino 1.59 **does not work with FastLED AT ALL**. Period. I tried the most basic example included with FastLED (Blink.ino) and it fails with simply the 'NEOPIXEL' LED strip type selected. It hangs somewhere inside:
`  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);`
It never returns from that function call.

I tried it with FastLED v3.6, v3.5, v3.4, and v3.3.3. They all fail.

Teensyduino 1.54 works just fine with FastLED. I tested it with v3.6 and v3.5.

PaulStoffregen, 2024-04-05:
fixed with https://github.com/PaulStoffregen/cores/commit/638a2079d290e3d6683aeceb07b6dfd8b09151a9

closed 2024-04-05 by PaulStoffregen