PaulStoffregen/cores #751: C++20 support (almost) working out of the box

firesurfer, 2024-08-22:
Hi I would like to use / I actually managed to use c++20 on a teensy4.1 with platform.io. 
I write this issue as there are only a few small adaptions needed in order to get it to work.

### 1. My platform.io configuration

```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
src_filter = +<**/*.cpp>
build_unflags = -std=c++14 -std=gnu++14 -std=gnu++17
build_flags = -std=c++20 -Llib -D USB_SERIAL -Wall -Wextra  -Wl,-Map=output.map -O2  
platform_packages = toolchain-gccarmnoneeabi@1.120301.0, framework-arduinoteensy@1.158.0
monitor_speed = 115200
```

What's needed here is:

* Select the newest available gcc
* undef the default c++14 and c++17 flags
* set the selected standard to `-std=c++20` 
* Explicitly link with contents of the `lib` subfolder: `-Llib`  - this is needed as the `libarm_cortexM7lfsp_math.a` is not provided by a newer compiler by default anymore

### 2. Replace all calls in WString.cpp to itoa/utoa with ltoa and ultoa

In the file `.platformio/packages/framework-arduinoteensy/cores/teensy/WString.cpp` the compiler will complain about the missing `itoa/utoa` function.
This makes sense as they are not actually part of the official C standard library. I replaced them with ltoa and ultoa which are actually available but it would probably make sense to replace them with some of the conversion functions that are actually guaranteed to be available by the standard: https://cplusplus.com/reference/cstdlib/

PaulStoffregen, 2024-08-22:
Which libraries and software projects have you tested?

firesurfer, 2024-08-22:
I am using it for my own project, there I am using QNEthernet, RingBuffer, and an I2C driver for a LM75.

PaulStoffregen, 2024-08-22:
Just so you know, PJRC is not planning to update the gcc toolchain or C++ dialect any time soon.  We just recently switched from gcc 5.4 to 11.3, and from C++14 to C++17.  With both changes, initial testing looked good, but then many less commonly used libraries and certain program broke.  Not planning to do that again so soon.

firesurfer, 2024-08-22:
Thats kind of a pity given that c++20 brings a lot of very useful changes especially for embedded environments. 

I think it should be up to the user to select which version of c++ is used. You can still default to c++17 but allow to select a newer dialect (just as I did in the platform.io config). If the core library supports c++20 it would already help quite a lot.

PaulStoffregen, 2024-09-08:
Closing this for now.  C++20 support is not coming anytime soon.

closed 2024-09-08 by PaulStoffregen