PaulStoffregen/cores #777: C++ 20 support issues

iontodirel, 2025-02-23:
I was able to successfully use C++ 20, by modifying the teensy\hardware\avr\1.59.0\boards.txt.

However, I noticed I am getting some linker errors about definitions not found for few C functions.

I fixed this using the following:

#include <sys/stat.h>
#include <stdlib.h>
#include <unistd.h>

void _exit(int status) { while (1) {} }
int _close(int file) { return -1; }
int _fstat(int file, struct stat *st) { st->st_mode = S_IFCHR; return 0; }
int _isatty(int file) { return 1; }
int _lseek(int file, int ptr, int dir) { return 0; }
int _read(int file, char *ptr, int len) { return 0; }
int _write(int file, char *ptr, int len) { return len; }
int _kill(int pid, int sig) { return -1; }
int _getpid(void) { return 1; }
int _sbrk(int incr) { return -1; }

Not sure why these functions are needed, or what are they referenced by, but this was the biggest issue for me when targeting C++20.

WestfW, 2025-02-23:
Typically, this happens when you link with newlib, because printf() is filesystem-based and wants to pull in functions needed to implement "files."  Arduino's SAM and SAMD cores also work around this using "stubs."

(it MIGHT mean that some unexpected code is calling printf(), like https://github.com/arduino/ArduinoCore-sam/issues/47 )

A-Dunstan, 2025-02-24:
At least some of these stubs are already implemented in this repo, e.g. https://github.com/PaulStoffregen/cores/blob/master/teensy4/startup.c#L708

so it would seem something isn't right with your build environment if they were reported as missing.

ssilverman, 2025-12-15:
@iontodirel did you use `-std=c++20` or `-std=gnu++20`?